ISS-Mimic/Mimic #382: Soyuz and Progress Vinyl Arrays

ISS-Mimic, 2023-05-12:
Can we get some soyuz and progress vinyl cuts for the new array pattern? - Pattern Attached

[image]

ISS-Mimic, 2023-05-12:
[image]

SpaceTowel, 2023-05-16:
> Can we get some soyuz and progress vinyl cuts for the new array pattern? - Pattern Attached
> 
> 
> 
> [image]
> 
> 

Need dimensions please

ISS-Mimic, 2023-05-16:
The attached SVG should natively have the correct dimensions - (should be 46.68 x 15.01 mm)

GodspeedDiscvry, 2023-05-19:
Two files - one for blue side - I think any smaller grids inside will be too hard to remove.  One for gold side if you want the white diamond pattern.
[image]
[image]
[image]