ISS-Mimic/Mimic #384: Update screens to kivy 2.2 and multiple display size functionality

ISS-Mimic, 2023-06-07:
- [ ] Control
- [ ] LED Screen
- [ ] Mimic
- [ ] Simulate
- [ ] Settings
- [ ] Orbit
- [ ] Robo
- [ ] CDH
- [ ] EVA
- [ ] ECLSS
- [ ] USOS
- [ ] VV
- [ ] Crew
- [ ] SCI
- [ ] ROS
- [ ] EPS
- [ ] C&T
- [ ] ISS
- [ ] TCS
- [ ] GNC

ISS-Mimic, 2023-06-07:
EVA screen needs text adjusting

GodspeedDiscvry, 2023-06-09:
Checking out on 7" and I have the following:

- giant text on Simulated screen for "OFT2 Loop" and "Disco!".  The disco one should be olkay - but will be a button - and actually OFT2 Loop will probably be a button too.
- orbit data the upper right text is now smaller and does not line up with identifiers
- orbit still does not work (no data displayed)
- US EVA we already discussed some issues here - but the crewlock status in upper right is not showing code that is in GUI instead of EVA screen code (and this was not actually on your list above)
- SGANT is doing something similar in that our transparent png is including a black box that wipes out other stuff (like the back button) when the transparent png rotates
---
Screens not done at the time if this comment:  Russian Docking, Robo, MT/MBS, SSRMS, SPDM, VV - will check those once checked off.  I did however include the C&T subs on my check even though not annotated above.

ISS-Mimic, 2023-06-09:
updates done and pushed - any new/continuing problems if you could individually make a new issue for each please

closed 2023-06-09 by ISS-Mimic