ISS-Mimic/Mimic #397: GUI screen not loading.  even after using just 'python GuI.py'

SpaceTowel, 2023-07-09:


GodspeedDiscvry, 2023-07-09:
Working fine for me.  Try timez.  Sometimes it will lock up if timez not done.

GodspeedDiscvry, 2023-07-09:
By the way, there are aliases for mpost things now.  "timez" for the sudo rdate thing, "mim" to change folders, "gui" once there to start gui.

GodspeedDiscvry, 2023-07-09:
Okay - sorry.  We have to add alias bash file to each mimic.  I opened a sep issue to possibly do this.  As for the main branch not working.... may need Sam on that one.  I will go look and see if I accidentally added one of the new screens to the main branch which could break it.

GodspeedDiscvry, 2023-07-09:
I did not see any new updates to anything on Main.  Chen says the one cloned Friday is fine, new one just shows white screen.

ISS-Mimic, 2023-07-09:
I bet its another consequence of the username not being "pi". Maybe looking for the screens in the wrong spot or something. Will need to find and protect against that too

ISS-Mimic, 2023-07-09:
Yep all the screens backgrounds are hard coded directories. will need some time to fix

ISS-Mimic, 2023-07-09:
Okay I think I fixed it all on the susan branch. @SpaceTowel if you could checkout the susan branch, and pull, and see if that works that would be good.

`git checkout susan`
`git pull`

ISS-Mimic, 2023-07-09:
also the telemetry has been offline since Thursday :(

ISS-Mimic, 2023-07-09:
Will need to merge susan branch for fixes, but probably ready to do that anyway

closed 2023-07-09 by ISS-Mimic