ISS-Mimic/Mimic #442: HiFI Robotic arm stil file need to upload to Github

SpaceTowel, 2023-11-04:
Upload the latest Robotic arm stl files to Github. reference to folder:   https://github.com/ISS-Mimic/Mimic/tree/main/3D_Printing/SSRMS_Canadarm_and_MT_MBS

HerculesFerrari, 2023-11-04:
I shared a MT converted to solid from a STL Sam found on CSA's site on discord not long ago. I'll look at what I have for SSRMS/CANADARM2/DEXTRE to get that ball rolling.

CSA's 3d assets don't have a license, but they do have usage terms to comply with copyright. IIRC it will just amount to attribution.

streadgold, 2023-11-04:
@HerculesFerrari I have the SSRMS ready (modified slightly to work with a dowel pin to swivel and connect to a mimic PDGF) but yeah DEXTRE is on the CSA website also and needs to be converted

streadgold, 2023-11-04:
As far as licensing goes we are fine with attribution 

"The CSA grants you permission to reproduce the materials on this site for personal, educational or informational non-commercial use, as indicated in the [Copyright Act](https://laws-lois.justice.gc.ca/eng/acts/C-42/index.html), so long as you meet the following conditions:

    You exercise due diligence in ensuring the accuracy of the materials reproduced.
    You acknowledge the CSA as the source of the material (""© Canadian Space Agency"", ""Credit: Canadian Space Agency"", ""@csa_asc"", etc.).
    You do not use the materials in such a way as to imply any association with or endorsement from the CSA.
    You do not use the materials in a deliberately misleading or defamatory context.
    You do not use the materials for partisan or political purposes

closed 2023-11-04 by ISS-Mimic