ISS-Mimic/Mimic #511: test new playback screen

streadgold, 2024-01-28:
@juggler2000  can you checkout the devsam branch and test the new playback screen on the franklin mimic? need to know if it works before I merge

juggler2000, 2024-01-28:
Yes, or @spacetowel

juggler2000, 2024-02-01:
Tested 1/29 and it worked.

ISS-Mimic, 2024-02-01:
I'm still seeing odd behavior on my setup. I'm not convinced this is ready to be merged

GodspeedDiscvry, 2024-02-15:
Walked through all functions - here are my notes :
(these are based on me not knowing that the speed selection was not coded yet)

- Should LEDs be changing during DISCO?  I think they are mosly white with some blue.
- If changing speed for same data dropdown, you must reselect the data in order for the START button to appear again.
- Speeds (from +/- buttons) are RANDOM for data drop dropdowns - like HTV is 32, 57, 61x...  And not the same across the board.
- For the most part buttons appeared to be close to where they should be.  +/- are roughly in right spot but do not look right at all compared to the graphics we have loaded on github.   They also seem way oversized - but I want to compare to the small screen and see if related to autoscaling.  
- I do need to tweak some of the sizing - like the Select on the data panel is too big.

juggler2000, 2024-02-23:
@GodspeedDiscvry , did you find out anything further on this?  Is there something I need to do?

PR #685 (2025-08-29): REFACTOR EVERYTHING
Fixes #665, Fixes #629, Fixes #611, Fixes #610, Fixes #598, Fixes #576, Fixes #511, Fixes #333, Fixes #321, Fixes #314, Fixes #201, Fixes #116, Fixes #53

Changes proposed in this request:
- Refactors the WHOLE project
- Every screen now keeps its logic on its own .py file
- GUI only contains functions that every screen uses
- Editing a screen only requires touching its .py and .kv files now, no changes to GUI

@ISS-Mimic