ISS-Mimic/Mimic #512: Update Mimic Frame CAD Model

streadgold, 2024-01-29:
Hey @Digisembler, can you update the aluminum frame model? The one we have in the integrated mimic model is definitely outdated 

[image]

[image]

The 6x side legs are definitely the wrong length, and the L-brackets are the small version when at least some of them should be the large version.

This is crucial since its the only reference for folks building their own

streadgold, 2024-01-31:
https://github.com/ISS-Mimic/Mimic/blob/main/Wiki/Mimic%20One%20Frame%20Updated.pptx

Digisembler, 2024-01-31:
Model can get updated - the powerpoint of the frame build is in the wiki under Mimic One Frame Updated.pptx

Digisembler, 2024-02-03:
So there are some issues with the tool i'm using for this - i cannot add the T plates and as far as i can tell, there's only the small 90 degree brackets.  We'll just have to point out which are which, or cad up those brackets from outside the tool.  Updated dimensions as best possible to the link above, realizing that everyone's specific build may have to adjust slightly for their build.  Step file included here to check out and receive comments back.
[UpdatedFrame240203.zip](https://github.com/ISS-Mimic/Mimic/files/14154288/UpdatedFrame240203.zip)

closed 2024-02-14 by streadgold