ISS-Mimic/Mimic #529: Check how new touch screen monitor looks / work (ASUS)

SpaceTowel, 2024-02-14:


GodspeedDiscvry, 2024-02-16:
Used new monitor to evaluate screen autoscaling.  Overall it appears that the background graphics and the overlaid graphics/text are scaling differently.

GodspeedDiscvry, 2024-02-16:
Examples of overlaid (coded) text not scaling properly:

![Resized_20240214_1
![Resized_20240214_16464
[image]
4](https://github.com/IS
[image]
S-Mimic/Mimic/assets/33506403/29c1a18c-5074-4fb3-aa18-927a16b099ee)
64804](https://github.com/ISS-Mimic/Mimic/assets/33506403/800e01f0-8874-4b8c-819e-09677358ffa0)
[image]
[image]

GodspeedDiscvry, 2024-02-16:
More examples of overlaid (coded) text not scaling properly.
[image]
[image]
[image]

GodspeedDiscvry, 2024-02-16:
Orbit Screen, however, is totally messed up.  
[image]