ISS-Mimic/Mimic #647: memory leak

imyxh, 2025-02-12:
Over a couple days this tab leaks a *lot* of memory:

<details>
<summary>about:memory</summary>

```
webIsolated=https://iss-mimic.github.io (pid 3742192)
Explicit Allocations

7,613.44 MB (100.0%) -- explicit
├──3,477.79 MB (45.68%) -- window-objects/top(https://iss-mimic.github.io/Mimic/, id=1241)
│  ├──3,064.90 MB (40.26%) -- active/window(https://iss-mimic.github.io/Mimic/)
│  │  ├──3,057.91 MB (40.16%) -- js-realm(https://iss-mimic.github.io/Mimic/)
│  │  │  ├──3,057.38 MB (40.16%) -- classes
│  │  │  │  ├──1,977.00 MB (25.97%) -- class(Object)/objects
│  │  │  │  │  ├──1,394.06 MB (18.31%) -- malloc-heap
│  │  │  │  │  │  ├──1,342.64 MB (17.64%) ── elements/normal
│  │  │  │  │  │  └─────51.42 MB (00.68%) ── slots
│  │  │  │  │  └────582.94 MB (07.66%) ── gc-heap
│  │  │  │  ├────613.32 MB (08.06%) ── class(NodeList)/objects/gc-heap
│  │  │  │  ├────135.64 MB (01.78%) -- class(Function)/objects
│  │  │  │  │    ├──135.57 MB (01.78%) ── gc-heap
│  │  │  │  │    └────0.07 MB (00.00%) ── malloc-heap/slots
│  │  │  │  ├────134.89 MB (01.77%) -- class(Call)/objects
│  │  │  │  │    ├──134.88 MB (01.77%) ── gc-heap
│  │  │  │  │    └────0.01 MB (00.00%) ── malloc-heap/slots
│  │  │  │  ├────104.67 MB (01.37%) -- class(Array)/objects
│  │  │  │  │    ├──104.65 MB (01.37%) ── gc-heap
│  │  │  │  │    └────0.03 MB (00.00%) ++ malloc-heap
│  │  │  │  └─────91.85 MB (01.21%) ++ (5 tiny)
│  │  │  └──────0.53 MB (00.01%) ++ (2 tiny)
│  │  └──────6.99 MB (00.09%) ++ (3 tiny)
│  └────412.89 MB (05.42%) -- js-zone(0x7a83ad342d00)
│       ├──340.78 MB (04.48%) ── compartments/cross-compartment-wrapper-tables
│       └───72.11 MB (00.95%) ++ (12 tiny)
├──3,223.75 MB (42.34%) ── heap-unclassified
├────799.37 MB (10.50%) -- js-non-window
│    ├──737.90 MB (09.69%) -- zones
│    │  ├──735.20 MB (09.66%) -- zone(0x7a83ad342d00)
│    │  │  ├──709.43 MB (09.32%) -- realm([Expanded Principal [https://iss-mimic.github.io/Mimic/, moz-extension://2e869e37-7697-4b4b-92c2-ad7b90a96733/]], Content Script "uBlock Origin" (ID: uBlock0@raymondhill.net, moz-extension://2e869e37-7697-4b4b-92c2-ad7b90a96733/))
│    │  │  │  ├──709.42 MB (09.32%) -- classes
│    │  │  │  │  ├──613.33 MB (08.06%) ── class(Proxy)/objects/gc-heap
│    │  │  │  │  ├───96.00 MB (01.26%) -- class(Array)/objects
│    │  │  │  │  │   ├──96.00 MB (01.26%) -- malloc-heap
│    │  │  │  │  │   │  ├──96.00 MB (01.26%) ── elements/normal
│    │  │  │  │  │   │  └───0.00 MB (00.00%) ── slots
│    │  │  │  │  │   └───0.00 MB (00.00%) ── gc-heap
│    │  │  │  │  └────0.09 MB (00.00%) ++ (2 tiny)
│    │  │  │  └────0.02 MB (00.00%) ++ sundries
│    │  │  └───25.77 MB (00.34%) ++ (4 tiny)
│    │  └────2.70 MB (00.04%) ++ (3 tiny)
│    └───61.46 MB (00.81%) ++ (3 tiny)
├────109.44 MB (01.44%) -- xpconnect
│    ├──109.39 MB (01.44%) ── runtime
│    └────0.04 MB (00.00%) ++ (5 tiny)
└──────3.10 MB (00.04%) ++ (21 tiny)

Other Measurements

4,774.70 MB (100.0%) -- decommitted
├──3,561.26 MB (74.59%) -- heap
│  ├──1,810.29 MB (37.91%) ── decommitted
│  └──1,750.96 MB (36.67%) -- unused-pages
│     ├──1,747.64 MB (36.60%) ── madvised
│     └──────3.33 MB (00.07%) ── fresh
└──1,213.44 MB (25.41%) ── js-non-window/gc-heap/decommitted-pages

861 (100.0%) -- event-counts
└──861 (100.0%) -- window-objects/top(https://iss-mimic.github.io/Mimic/, id=1241)/active/window(https://iss-mimic.github.io/Mimic/)/dom
   ├──750 (87.11%) ── event-listeners
[... 167 lines cut ...]
     0.00 MB ── imagelib-surface-cache-estimated-locked
     0.00 MB ── imagelib-surface-cache-estimated-total
           0 ── imagelib-surface-cache-image-count
           0 ── imagelib-surface-cache-image-surface-count
           0 ── imagelib-surface-cache-locked-image-count
           0 ── imagelib-surface-cache-locked-surfaces-count
           0 ── imagelib-surface-cache-overflow-count
           0 ── imagelib-surface-cache-table-failure-count
           0 ── imagelib-surface-cache-tracked-cost-count
           0 ── imagelib-surface-cache-tracked-expiry-count
           0 ── imagelib-surface-cache-tracking-failure-count
     0.78 MB ── js-main-runtime-temporary-peak
   5,943,850 ── page-faults-hard
  22,758,853 ── page-faults-soft
 4,921.97 MB ── resident
 7,394.30 MB ── resident-peak
 4,886.61 MB ── resident-unique
     0.00 MB ── script-preloader-memmapped-cache
     0.00 MB ── system-heap-allocated
13,283.00 MB ── vsize
     0.00 MB ── wasm-runtime

End of webIsolated=https://iss-mimic.github.io (pid 3742192)
```
</details>

SWGlassPit, 2025-02-12:
This is unfortunately a known condition and as best we can tell, it seems to be in the library code. The code on our end doesn't make any explicit allocations.

SWGlassPit, 2025-02-12:
At the moment my best recommendation would be to refresh the tab once a day or so if the intent is to keep it running continuously. Unfortunately the information I have available to me doesn't really indicate where it's happening other than somewhere on this page. 

I do know that when a tab gets backgrounded, the events pile up, so some sort of onBlur type of code would be good to include to shut down the listener when the tab isn't visible. I don't know if this would help this particular problem though, and I haven't had time to explore this.