ISS-Mimic/Mimic #648: USLAB000ALT - ISS Altitude from State Vectors

Zamo1963, 2025-02-28:
This value is not present in the list of feeds, can you add it?

ISS-Mimic, 2025-02-28:
What list of feeds? So the altitude is not actually a telemetry item, its just calculated from the state vector values and displayed

Zamo1963, 2025-02-28:
This list:
https://github.com/ISS-Mimic/Mimic/blob/main/Telemetry/iss_telemetry_all.js

Zamo1963, 2025-02-28:
https://github.com/ISS-Mimic/Mimic/blob/main/Telemetry/iss_telemetry_all.js

SWGlassPit, 2025-02-28:
Yeah that's a post-calculated value and isn't a direct telemetry feed, so it would not be appropriate to add to the js file you listed, which is for the explicit purpose of setting up all the lightstreamer subscriptions. You can see how it's calculated from the page source of the page you screenshotted.

closed 2025-02-28 by SWGlassPit